feaser/microtbx-modbus #2: Baudrate 115200 does not work

JannickIDA, 2024-07-08:
Hello, im using this library as a modbus client with freertos. 115200 baudrate is not working. i tried some debugging myself and within "TbxMbRtuCreate" the 1.5 and 3.5 character times are only calcutud to 19200. when i tried to calculate this it worked with superloop but occasionally i got an timeout and with freertos continously an timeout. i tested with an usb to rs485 adapter and the messages seems fine.

feaser, 2024-07-10:
Hello Jannick,

Thank you for your interest in MicroTBX-Modbus.

According to the Modbus protocol specification, a fixed time should be used for the 1.5 and 3.5 character timeouts, for baudrates higher than 19200 bps. That's why `TbxMbRtuCreate()` only calculates them up to 19200 bps.

When 115200 bps is not working, this is typically caused by something in your port. To debug this, I would first set a breakpoint in your UART interrupt handler. Next, send a byte from your PC at 115200 bps and verify that the UART interrupt handler gets triggered for a byte reception event.

If the byte reception interrupt did not occur, then something is not yet right in your `TbxMbPortUartInit()`. It could be as simple as a clock prescaler configuration. Once the byte reception is working at 115200 bps, you know that the baudrate configuration is all set. Byte transmit then typically also works, so no need to verify that really.

If at this point you still run into communication problems, it is likely caused by your `TbxMbPortTimerCount()` implementation. This function should return a counter value that increments every 50 microcseconds. You can test this by adding some temporary testcode to your application, which first calls `TbxMbPortTimerCount()`, next delays for 5 milliseconds, and then calls `TbxMbPortTimerCount()` one more time. The values returned by the two calls to `TbxMbPortTimerCount()` should have a delta of about 5000us / 50us = 100.

Good luck and with kind regards,

Frank

closed 2024-07-10 by feaser

JannickIDA, 2024-07-10:
Hello Frank,

Thank you for your response.

I made a logger in "HAL_UART_RxCpltCallback" in "tbxmb_port.c" where i log each byte recieved. and all bytes i expect are present in the correct order. HAL_UART_GetError does not return any errors so "TbxMbUartDataReceived(TBX_MB_UART_PORT1, &rxByte, 1U);" is executed each byte recieved.

Byte transmit does definitely work since my modbus slave recieves the correct data on 115200bps from my end.

my apb timer clocks run on 84mhz so i have a prescaler of 4199. I did the test with the 5ms delay and my delta varies between 101 and 119 avaraging on about 104-105. so this might by a little to high?

for context on 38400bps 4/5 messages end in timeout and on 115200 all messages. in"tbxmb_clinet.c" "TbxMbClientTransceive" returns TBX_ERROR on line 276.

Kind regards,

Jannick van de Riet