feaser/microtbx-modbus #3: Program hangs in tbxmb_event.c while loop

sonarus-mcu, 2024-12-28:
Hello,

First of all, thank you for creating such a great project! I am using microtbx-modbus in my project on STM32H743, and it has been really helpful so far.

However, I've encountered an issue where the program hangs during the execution of the while loop in the tbxmb_event.c file:
```ruby
while (listItem != NULL)
{
  /* Convert the opaque pointer to the event context structure. */
  tTbxMbEventCtx * eventPollCtx = (tTbxMbEventCtx *)listItem;
  /* Call its poll function if configured. */
  if (eventPollCtx->pollFcn != NULL)
  {
    eventPollCtx->pollFcn(listItem);
  }
  /* Move on to the next item in the list. */
  listItem = TbxListGetNextItem(pollerList, listItem);
}
```
At some point, the program gets stuck in this loop, and listItem seems to never become NULL, causing the program to freeze.
I would appreciate any guidance or help in identifying the root cause of this issue and how I can fix it.
Thank you for your time and support!

feaser, 2025-01-07:
Hi,

Thank you for the kind words and your interest in MicroTBX-Modbus.

The code snippet you posted from `TbxMbEventTask()` iterates over a linked list until the end is reached. For this loop to get stuck, something would need to keep adding items to the list. However the only function adding something to this list is `TbxMbEventTask()` itself. 

With other words, that situation can only happen if somehow `TbxMbEventTask()` gets called again, while it is already running. For example if you have multiple RTOS tasks calling `TbxMbEventTask()` or if it's called by an interrupt. 

You could check if this is the case in your application and correct it. In a traditional superloop setup, `TbxMbEventTask()` should be called continuously in the superloop. If you use an RTOS, instead of a superloop, make sure `TbxMbEventTask()` is only called from one task.

If that doesn't resolve the issue, look at things such as (task) stack overflow.

closed 2025-03-24 by feaser

feaser, 2025-04-28:
I was able to reproduce the problem where the program gets stuck in the while-loop in `TbxMbEventTask()`. 

The fix for this issue is available in MicroTBX-Modbus release 0.9.4.