feaser/microtbx-modbus #4: Program stuck in while loop

AMARNATHMANGALI, 2025-04-22:
Hello,

I hope you're doing well, and thank you for providing such a great and helpful project. I’m currently integrating MicroTBX-Modbus into my application on the MIMXRT1176 controller.

I’ve encountered an issue where the application appears to hang during execution of the following loop in the tbxmb_event.c file, specifically within the TbxMbEventTask() function:

void * listItem = TbxListGetFirstItem(pollerList);
while (listItem != NULL)
{
  tTbxMbEventCtx * eventPollCtx = (tTbxMbEventCtx *)listItem;

  if (eventPollCtx->pollFcn != NULL)
  {
    eventPollCtx->pollFcn(listItem);
  }

  listItem = TbxListGetNextItem(pollerList, listItem);
}

**Observations:**
The loop seems to never exit, causing the system to freeze.

The TbxMbEventTask() function is **called only from** TbxMbOsalSemTake().

While TbxMbOsalSemTake() is used in multiple places, the call chain always originates from it.

The Modbus use case in my application involves reading holding registers every 30–50 ms.

**Request for Guidance:**
Based on previous suggestions, I understand this could be due to reentrant calls or concurrency issues. However, since TbxMbEventTask() is not called directly elsewhere, I would appreciate further clarification or recommendations on how to resolve this.

Thank you for your time and support.

Best regards,
Amarnath
```

AMARNATHMANGALI, 2025-04-22:
[image]

AMARNATHMANGALI, 2025-04-22:
[image]

Complete call heirarchy.

feaser, 2025-04-22:
Hi Amarnath,

In your description you state that the `TbxMbEventTask()` function is called only from `TbxMbOsalSemTake()`. If that it truly the case, then MicroTBX-Modbus is not yet properly integrated:

- If you don't use an RTOS in your system, make sure to compile and link `tbxmb_superloop.c` with your project and call `TbxMbEventTask()` continuously from program's infinite superloop.
- If you use an RTOS in your system (FreeRTOS is currently supported), make sure to compile and link `tbxmb_freertos.c` with your project, instead of `tbxmb_superloop.c`. Then create a new task and call `TbxMbEventTask()` continuously from the task's infinite loop.

An example is available in the [getting started section](https://feaser.github.io/microtbx-modbus/gettingstarted/) of the online user manual.

If the problem still persists, you can try to:

- Change the task priority of your Modbus task, assuming that you use FreeRTOS.
- Verify that there are not stack overflow issues.
- Disable the data and instruction cache of the Cortex CPU.

Good luck!

AMARNATHMANGALI, 2025-04-22:
Hi @feaser as per your suggestion I have added tbxmb_freertos.c by removing the superloop.c. I am calling the TbxMbEventTask() function in my FreeRTOS thread (only in 1 thread). I am using my modbus as **client**. However , I tried to read the holding registers . But continuously I am getting the Txpacket as NULL.

uint8_t TbxMbClientReadHoldingRegs(tTbxMbClient   channel,
                                   uint8_t        node,
                                   uint16_t       addr,
                                   uint8_t        num,
                                   uint16_t     * holdingRegs)
{
  uint8_t result = TBX_ERROR;

  /* Verify the parameters. */
  TBX_ASSERT((channel != NULL) && (node <= TBX_MB_TP_NODE_ADDR_MAX) && (num >= 1U) &&
             (num <= 125U) && (holdingRegs != NULL));

  /* Only continue with valid parameters. */
  if ((channel != NULL) && (node <= TBX_MB_TP_NODE_ADDR_MAX) && (num >= 1U) &&
      (num <= 125U) && (holdingRegs != NULL))
  {
    /* Convert the client channel pointer to the context structure. */
    tTbxMbClientCtx * clientCtx = (tTbxMbClientCtx *)channel;
    /* Sanity check on the context type. */
    TBX_ASSERT(clientCtx->type == TBX_MB_CLIENT_CONTEXT_TYPE);

    /* Obtain write access to the request packet. */
    tTbxMbTpPacket * txPacket = clientCtx->tpCtx->getTxPacketFcn(clientCtx->tpCtx);
    /* Should always work, unless this function is being called recursively. Only
     * continue with access for preparing the request packet.
     */
    **if (txPacket != NULL)**
    {
      /* Prepare the request packet. */
      txPacket->node = node;
      txPacket->pdu.code = TBX_MB_FC03_READ_HOLDING_REGISTERS;
      txPacket->dataLen = 4U;
      /* Starting address. */
      TbxMbCommonStoreUInt16BE(addr, &txPacket->pdu.data[0]);
      /* Number of registers. */
      TbxMbCommonStoreUInt16BE(num, &txPacket->pdu.data[2]);

      /* Determine the request type (broadcast / unicast). */
      uint8_t isBroadcast = TBX_FALSE;
      if (node == TBX_MB_TP_NODE_ADDR_BROADCAST)
      {
        isBroadcast = TBX_TRUE;
      }
      /* Transmit the request and wait for the response to a unicast request to come in
       * or the turnaround time to pass for a broadcast request.
       */
      result = TbxMbClientTransceive(clientCtx, isBroadcast);

      /* Only continue with processing the response if all is okay so far and the request
       * was unicast.
       */
      if ((result == TBX_OK) && (isBroadcast == TBX_FALSE))
      {
        /* Obtain read access to the response packet. */
        tTbxMbTpPacket * rxPacket = clientCtx->tpCtx->getRxPacketFcn(clientCtx->tpCtx);
        /* Since we just received a response packet, the packet access should always 
         * succeed. Sanity check anyways, just in case.
         */
        TBX_ASSERT(rxPacket != NULL);
        /* Only continue with packet access. */
        if (rxPacket != NULL)
        {
          /* Check that the response came from the expected node, that it's a response
           * with the same function code (not an exception response) and that the data
           * length and the byte count are as expected.
           */
          uint8_t byteCount = rxPacket->pdu.data[0];
          if ((rxPacket->node != node) ||
              (rxPacket->pdu.code != TBX_MB_FC03_READ_HOLDING_REGISTERS) ||
              (byteCount != (num * 2U)) ||
              (rxPacket->dataLen != (byteCount + 1U)) )
          {
            result = TBX_ERROR;
          }
          /* Response content valid. Process its data. */
          else
          {
            /* Set pointer to where the holding registers start in the response. */
            uint8_t const * regValPtr = &rxPacket->pdu.data[1];
            /* Read out and store the holding register values. */
            for (uint8_t idx = 0U; idx < num; idx++)
            {
              holdingRegs[idx] = TbxMbCommonExtractUInt16BE(&regValPtr[idx * 2U]);
            }
          }
        }
        /* Could not access the response packet. */
        else
        {
          result = TBX_ERROR;
        }
        /* Inform the transport layer that were done with the rx packet and no longer
         * need access to it.
         */
        clientCtx->tpCtx->receptionDoneFcn(clientCtx->tpCtx);
      }
    }
  }
  /* Give the result back to the caller. */
  return result;
} /*** end of TbxMbClientReadHoldingRegs ***/

I tried to debug the callback function.

static tTbxMbTpPacket * TbxMbRtuGetTxPacket(tTbxMbTp transport)
{
  tTbxMbTpPacket * result = NULL;

  /* Verify parameters. */
  TBX_ASSERT(transport != NULL);

  /* Only continue with valid parameters. */
  if (transport != NULL)
  {
    /* Convert the TP channel pointer to the context structure. */
    tTbxMbTpCtx * tpCtx = (tTbxMbTpCtx *)transport;
    /* Sanity check on the context type. */
    TBX_ASSERT(tpCtx->type == TBX_MB_RTU_CONTEXT_TYPE);
    /* Access to the transmission packet by a channel is only allowed outside the 
     * TRANSMISSION state. In this state the transmission path is locked until a
     * transition back to IDLE state is made. This happens once the transport layer
     * completed the packet transmission.
     */
    TbxCriticalSectionEnter();
    uint8_t currentState = tpCtx->state;
    TbxCriticalSectionExit();
    **if (currentState != TBX_MB_RTU_STATE_TRANSMISSION)
    {
      /* Update the result. */
      result = &tpCtx->txPacket;
    }**
  }
  /* Give the result back to the caller. */
  return result;
} /*** end of TbxMbRtuGetTxPacket ***/

Here My current state is always showing in the transmission state. So I was unable to enter into that update result part.

Please look into my issue.

Thank you for your time and support.

closed 2025-04-22 by feaser

feaser, 2025-04-28:
I was able to reproduce the problem where the program gets stuck in the while-loop in `TbxMbEventTask()`. 

The fix for this issue is available in MicroTBX-Modbus release 0.9.4.

AMARNATHMANGALI, 2025-04-28:
Thank you so much sir.