feaser/microtbx-modbus #5: Code stucks in ListItem loop

dominikstupka, 2025-04-22:
Hello Frank,

First of all, thanks for this awesome library and its clean design. Over the last few days I’ve been fighting an issue I can’t resolve. I’m running three devices on my Modbus RTU network: one client and two servers. The client is an ESPHome device, and both servers use your library on top of FreeRTOS.

Everything works fine for hours, sometimes days, but after a while something goes wrong. Under normal operation, pollerList oscillates between 0 and 1 items, but after some time its size jumps to 2 items:

[image]

After that, the code gets stuck inside this loop:
```c
  while (listItem != NULL)
  {
    /* Convert the opaque pointer to the event context structure. */
    tTbxMbEventCtx * eventPollCtx = (tTbxMbEventCtx *)listItem;
    /* Call its poll function if configured. */
    if (eventPollCtx->pollFcn != NULL)
    {
      eventPollCtx->pollFcn(listItem);
    }
    /* Move on to the next item in the list. */
    listItem = TbxListGetNextItem(pollerList, listItem);
  }
 ```
New interrupts still arrive from USART, but the code never leaves the loop. I’ve also monitored the physical bus with a logic analyzer and saw nothing abnormal. I can’t determine the root cause of this behavior.

Can you help me debug this issue?

Thanks a lot!

dominikstupka, 2025-04-24:
Hi, just a quick update — a few minutes after my last post yesterday, the code got stuck again just like before. So I went ahead and implemented your suggested code changes.

Right now, after 24 hours, everything looks good — no program freeze so far. I’ll let you know in a few days if everything continues to run smoothly.

Thanks again for your help!

dominikstupka, 2025-04-27:
Hi, another quick update — after 4 days of running, no issues or freezes have occurred.
It seems that this version of the code is running stable without any problems. Thanks a lot!

feaser, 2025-04-28:
Thank you for the update. Glad to hear that it's now working for you. I investigated the issue and workaround a bit futher, because it puzzled me why the while-loop has an issue. It turns out that the proposed work around works in your case, but another small change is needed to properly resolve this issue.

I found a way to reproduce the problem on my system. It is caused by two (or more) identical entries in the `pollerList`. This also explains why you noticed that the issue occurred, when the list size went from 1 to 2.

When two or more identical entries are in the `pollerList`, and you get to the while-loop, the first iteration uses the first item return by `TbxListGetFirstItem()`. At the end of the first iteration, the call to `TbxListGetNextItem()`, returns the second item, which will be used during the second loop iteration. So far so good.

Now at the end of the second loop iteration, the problem occurs. `TbxListGetNextItem()` is expected to return `NULL` because it is now at the end of the list. But `TbxListGetNextItem()` actually uses the item as a reference to determine the next item, and not its internal node in the linked list. Since item one and two are identical, `TbxListGetNextItem()` returns yet again the second item and not the assumed `NULL` to indicate the end of the list. Consequently, the while-loop keeps going.

The proper fix to this issue is to make sure a duplicate entry never makes it into the list. This is done by simply calling `TbxListRemoveItem()` before adding a new entry with `TbxListInsertItemBack()` inside function `TbxMbEventTask()`:

Original:

```c
        case TBX_MB_EVENT_ID_START_POLLING:
        {
          /* Poller list entries are allocated from a memory pool. This means that there
           * is no need to worry about heap fragmentation. Just make sure to cap the
           * maximum number of entries to prevent heap exhaustion.
           */
          TBX_ASSERT(TbxListGetSize(pollerList) <= TBX_MB_EVENT_QUEUE_SIZE);
          /* Only continue if the current poller list size is not yet maxed out. */
          if (TbxListGetSize(pollerList) <= TBX_MB_EVENT_QUEUE_SIZE)
          {
            /* Add the context at the end of the event poller list. */
            uint8_t insertResult = TbxListInsertItemBack(pollerList, newEvent.context);
            /* Check that the item could be added to the queue. If not, then the heaps size
             * is configured too small. In this case increase the heap size using
            * configuration macro TBX_CONF_HEAP_SIZE. 
            */
            TBX_ASSERT(insertResult == TBX_OK);
          }
        }
        break;
```
Fix:

```c
        case TBX_MB_EVENT_ID_START_POLLING:
        {
          /* Poller list entries are allocated from a memory pool. This means that there
           * is no need to worry about heap fragmentation. Just make sure to cap the
           * maximum number of entries to prevent heap exhaustion.
           */
          TBX_ASSERT(TbxListGetSize(pollerList) <= TBX_MB_EVENT_QUEUE_SIZE);
          /* Only continue if the current poller list size is not yet maxed out. */
          if (TbxListGetSize(pollerList) <= TBX_MB_EVENT_QUEUE_SIZE)
          {
            /* Delete first to prevent a duplicate entry. */
            TbxListRemoveItem(pollerList, newEvent.context);  /* <<<<<<<<<<<<<< ADDED <<<<<<<<<<<<<< */
            /* Add the context at the end of the event poller list. */
            uint8_t insertResult = TbxListInsertItemBack(pollerList, newEvent.context);
            /* Check that the item could be added to the queue. If not, then the heaps size
             * is configured too small. In this case increase the heap size using
            * configuration macro TBX_CONF_HEAP_SIZE. 
            */
            TBX_ASSERT(insertResult == TBX_OK);
          }
        }
        break;
```
 
I'll go ahead and process this fix. I'll also keep the change from the while-loop to the if-loop. It's not really needed anymore, yet still strikes me as the safer way to iterate the `pollerlist` in `TbxMbEventTask()`.

feaser, 2025-04-28:
The fix for this issue is available in MicroTBX-Modbus release 0.9.4.

closed 2025-04-28 by feaser